rusefi/hellen88bmw-issues #2: Pin 2 - ICV close (currently NC)

OrchardPerformance, 2021-08-17:
Pin 2 should be the close winding of the ICV, we support 2 wire mode so we should probably use it and not be using resistors or any hacks to run the ICV

rusefillc, 2021-08-19:
@OrchardPerformance is this the dual-solenoid idle valve like subaru?

@ivanchuba need same high-current MOSFET and same diode like the other channel of idle solenoid

rusefillc, 2021-08-31:
@ivanchuba need same high-current MOSFET and same diode like the other channel of idle solenoid

so in total please manually add two ZXMS6005SGTA same as Q312

ivanchuba, 2021-08-31:
[image]
done

Qwerty-OFF, 2021-09-15:
@OrchardPerformance please close ticket if that's what we need
[image]