rusefi/hellen88bmw-issues #3: Pin 8 Check engine light (currently NC)

OrchardPerformance, 2021-08-17:
We should connect the check engine light as the software will use the light in future to warn of errors. (it is in the smart ECU plans).

andreika-git, 2021-08-19:
@OrchardPerformance Is CE lamp lowside-active or highside-active?

rusefillc, 2021-08-24:
@ivanchuba please add a via for now

ivanchuba, 2021-08-28:
> @OrchardPerformance Is CE lamp lowside-active or highside-active?

@OrchardPerformance not have answer?

OrchardPerformance, 2021-08-28:
Will try to check when its daylight tomorrow

ivanchuba, 2021-08-28:
@OrchardPerformance ok, thanks

Qwerty-OFF, 2021-09-15:
@OrchardPerformance Did you check how to connect the lamp?

rusefillc, 2022-06-20:
@Qwerty-OFF do we have at least a via?

Qwerty-OFF, 2022-06-20:
Via in rev. a
Connected now:
[image]

closed 2022-06-20 by Qwerty-OFF