rusefi/hellen88bmw-issues #5: Double check MAF meter pins 14 and 41

OrchardPerformance, 2021-08-17:
I have the MAF working connected to my frankenso, let me double check what I wired to each pin before we order. 
If I forget you have permission to phone me and make me go do it.

rusefillc, 2021-08-24:
41 we have jumper we are good to fix if needed

[image]

@ivanchuba we need jumper on 14 just in case 
[image]

@ivanchuba please add jumper and assign back to @OrchardPerformance

ivanchuba, 2021-08-28:
done, please check

ivanchuba, 2021-08-31:
[image]

Qwerty-OFF, 2021-09-15:
Now we can use pins 14 or 41. @OrchardPerformance please close ticket if this is final solution for rev. a