rusefi/hellen88bmw-issues #10: M50 cam sensor difference

OrchardPerformance, 2021-08-18:
Non VVT (non Vanos) M50 engines with the M3.1 ECU are 88 pin and use a VR sensor for the cam.
VVT/VANOS M50TU uses the M3.3.1 ECU with 88pin and a Hall sensor. 
The are nearly perfectly pin compatible but this could be a problem if we do not allow for it in the design.

rusefillc, 2021-08-24:
I see jumpers on relevant pins we can probably survive first engine start

[image]

[image]

Qwerty-OFF, 2021-09-15:
We can use second channel in VR module for this