rusefi/hellen88bmw-issues #11: Pin 46 - TI output

OrchardPerformance, 2021-08-18:
This is the fuel consumption output to get the dash econometer and the on board computer to show consumption and range correctly. 
I believe we will be able to provide this signal but we currently have nothing in software. 
I think the signal is a 12v pulse equal to the pulse width with a small modification for dead time. It is only accurate for the standard injectors for the car. 
We will need to be able to adjust the pulse width by a factor to correct the signal for different injectors.

OrchardPerformance, 2021-08-18:
I believe this requires a simple lowside

rusefillc, 2021-08-19:
just a via for now and keep ticket open

ivanchuba, 2021-08-31:
[image]
done