rusefi/hellen88bmw-issues #13: Check BMW relay config

OrchardPerformance, 2021-08-18:
If wired correctly the ECU will power up as OEM via the DME relays, if not it will get stuck on. 
Task here to review +12v connections against BMW wiring diagrams.

OrchardPerformance, 2021-08-18:
Pin 26 - 12v at all times, large red wire
Pin 27 - Switched lowside for main relay activation, small brown wire
Pin 54 - 12v switched by main relay, large red/white wire
Pin 56 - 12v switched by ignition barrel 

Red/White wires power the injectors, vvt ect from the main relay. 
Red wires are 12v permanent on, direct connection to battery through main fuse
Green wires are 12v switched by ignition switch
Brown is GND 

TLDR:
RE should only be powered by the Green wire on Pin 56 and this should provide the main on/off power. 
Any secondary items powered from the ECU like the high sides should probably be powered from the red/white on pin 54.
If we wanted keep alive memory then power for that is on Pin 26.

rusefillc, 2021-08-19:
нужно пока заменить джампер JP54 на "DNP" (и тикет пока не закрывать)

ivanchuba, 2021-08-23:
done

rusefillc, 2021-08-24:
@andreika-git @stefanst can you please validate? that's a sore and important subject