rusefi/hellen88bmw-issues #16: MAIN_RELAY is totally *ed up!

andreika-git, 2021-08-19:
[image]
[image]
[image]
[image]

rusefillc, 2021-08-19:
@ivanchuba please elaborate on what went wrong here. We've touched this area twice already, where is the misunderstanding happening - is the problem with the wiring diagram? is the problem with naming on the modules? else?

rusefillc, 2021-08-19:
Подключение главного реле перепутано напрочь:
- "MAIN_RELAY" и "OUT_MAIN" поменялись местами
- выходы на главное реле замыкаются на входы
- почему-то подключаются к шинам зажигания и инжекторов, что ещё больше запутывает

Как должно быть сделано правильно:
- выход модуля mcu OUT_IO1 - это MAIN_RELAY
- MAIN_RELAY идёт на LOW5_MAIN модуля output
- выход OUT_LOW5_MAIN модуля output - это OUT_MAIN
- OUT_MAIN идёт на пин 27 коннектора через джампер

ivanchuba, 2021-08-23:
done

closed 2021-08-23 by ivanchuba