rusefi/hellen88bmw-issues #18: Solenoid flyback diodes? Idle solenoid?

andreika-git, 2021-08-19:
Do we have at least one solenoid in this car? Maybe the idle valve?
Then we need flyback diodes for them like on the other boards:
[image]
Currently I don't see any of those on the board.

andreika-git, 2021-08-19:
Maybe we should also use a separate power MOSFET for idle actuator as in HellenNB2?

rusefillc, 2021-08-19:
@andreika-git do we have documentation for current limits of 

[image]

Shall we rename OUT_LOW1 to OUT_LOW1_RELAY

andreika-git, 2021-08-19:
https://github.com/andreika-git/hellen-one/wiki/module-output

ivanchuba, 2021-08-28:
done, please check

rusefillc, 2021-08-29:
Давай поставим наши любимые B5819W (C8598) 1A rated?

rusefillc, 2021-08-31:
B5819W and  separate power MOSFET like https://github.com/rusefi/rusefi/wiki/Hardware/Hellen/hellen72-d-schematic.pdf same as on injector output channels just manual schematics

ivanchuba, 2021-08-31:
[image]
done