rusefi/hellen88bmw-issues #34: All DNP jumpers are populated

stefanst, 2021-10-30:
JP6, JP 20, JP 21, JP 41 and JP 56 should be DNP as per the schematic. They are, in fact, populated though.
[image]
[image]
[image]
[image]

rusefillc, 2021-11-28:
https://github.com/andreika-git/hellen-one/issues/78

Qwerty-OFF, 2022-06-21:
Fixed

closed 2022-06-21 by Qwerty-OFF