rusefi/hellen88bmw-issues #37: Injector channels not firing

OrchardPerformance, 2021-11-07:
I cannot seem to detect any injector outputs triggering while the board is in self stim mode. 

I have probed around with the oscilloscope and there seems to be no output but it hard to be sure I am probing the correct pins to be sure. 
Schematic says pins are E25, 26, 29, 30, 31, 32, 33, 34. 
It is not clear exactly which pins that is on the module.

rusefillc, 2021-11-07:
@OrchardPerformance tunes are useful for troubleshooting 
@OrchardPerformance logs are also pretty useful

OrchardPerformance, 2021-11-07:
Closing this as it seems the drivers only fire when there is 12v supply. 

Leaving this as a reminder to put a note into the documentation that bench testing with only 5v supply will not allow injector drivers to be tested.

closed 2021-11-07 by OrchardPerformance