rusefi/hellen88bmw-issues #50: Trigger loss at 6k or 6.6k on dyno

rusefillc, 2022-07-01:
One run we got trigher errors at 6600

Majority of runs we get constant trigher error at 6k

That's running crank only on max9926

mck1117, 2022-07-01:
post TS trigger log at idle

rusefillc, 2022-07-01:
tune https://rusefi.com/online/view.php?msq=1097

[e36 triger log idle 2022-07-01_13.19.16.zip](https://github.com/rusefi/hellen88bmw-issues/files/9030646/e36.triger.log.idle.2022-07-01_13.19.16.zip)

mck1117, 2022-07-01:
there's some setting like "show logic levels" in the trigger config, can you toggle that and grab another?

mck1117, 2022-07-01:
Turn if off then? That's why I said toggle.

rusefillc, 2022-07-01:
Sorry I misread that :(

rusefillc, 2022-07-10:
@stefanst also https://github.com/rusefi/rusefi/wiki/Trigger-Configuration-Guide