rusefi/hellen88bmw-issues #54: Populate new boards

stefanst, 2022-08-10:
This is the status of the new boards I received. I assume they are Rev B.
[image]
What parts do I need to add at this point? 
I know of:
- Injector drivers
- Igniters
- MAP sensor
What else should be populated and where does it go? I do not have access to secret Hellen Modules layout, so if Hellen modules need to be completed, I need to be told explicitly what to do, or where to find the information.
Any other mods to the boards before testing on JimStim?

rusefillc, 2022-08-10:
(Injector drivers + Igniters + MAP sensor + MAX9926 if plan to use it) seems like the complete list!

and https://github.com/rusefi/hellen88bmw-issues/issues/52 covers the analog VR side of fun

rusefillc, 2022-08-10:
PS: I have forgotten about U7 TLS115, did I by any chance ship those? that's for +5v sensor supply